linux4sam/egt #3: Eclipse IDE Setup is irreproducible

zborowa, 2020-01-31:
System information:
OS: CentOS7-1908
Kernel: 3.10.0-1062.9.1.el7.x86_64

I'm currently trying to reproduce your [Eclipse IDE Setup guide](https://linux4sam.github.io/egt-docs/eclipse.html) but it seems that the guide or my setup is missing some components.

1. During the building step of "Start Eclipse and Create EGT project" I'm getting the following errors:
```
/usr/include/c++/4.8.2/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support is currently experimental, and must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support for the \
  ^
```

In the current version of the Eclipse IDE, 2019-12, and the current version, 2019-06, there is no option to be found to set the g++ parameter for the "-std=c++11" flag. Is there some way to do this which is not specified in the guide?

and:

```
/buildroot-at91/output/host/bin/../arm-buildroot-linux-gnueabihf/sysroot/usr/include/gnu/stubs.h:7:29: fatal error: gnu/stubs-soft.h: No such file or directory
 # include <gnu/stubs-soft.h>
                             ^
compilation terminated.
make: *** [basic-basic.o] Error 1
"make all" terminated with exit code 2. Build might be incomplete.
08:45:56 Build Failed. 3 errors, 0 warnings. (took 307ms)
```

I skipped then all the remote debugging chapters and went to "Build libegt on Host Machine":
2. When adding a new project with the name "libegt" in the cloned directory "egt" Eclipse does not allow you to finish the setup because of an overlapping project called "egt". Does the project name matter? 

3. I also went through the README of [this repo](https://github.com/linux4sam/egt/blob/master/README.md) but during the `make` command after autogen.sh and configure the make errors on `unrecognized command line option '-std=c++14'` while I have nowhere configure it to use 14 or any other version.

joshua-henderson, 2020-02-18:
@zborowa Just wanted to let you know that this isn't being ignored.  We're going to take a deeper look at this tutorial part of the manual and see if it can be made more straightforward.

zborowa, 2020-02-18:
@joshua-henderson thanks for the notice. Would it be possible to help me with the errors? I've been busy with this for a while and I'm considering switching to something else if I can't get this to work this week.

SandeepSM, 2020-03-10:
@zborowa what is version of c++ compiler ? looks like OS: CentOS7-1908 is using old version of gcc (4.8.2) that doesn't support C++ 11 features. Please update the compiler.

joshua-henderson, 2020-05-08:
The eclipse section of the programmer's manual has been reviewed and tested for the up and coming 1.0 release.  If there is still an issue, feel free to re-open or create another issue.

closed 2020-05-08 by joshua-henderson