linux4sam/egt #4: Cross compiling from x886 Linux to ARM for Debian (no Buildroot)

amontefusco, 2020-06-02:
I'm trying to cross-compile the EGT from Linux Ubuntu 18.04, but without using Buildroot (as my target system runs a regular Debian distro for ARM.
On the same build platform, the build for x86 / X11 is working fine and all examples work.

```
./autogen.sh 
./configure --host=arm-linux-gnueabi  --without-x11
make -j4
```

But I get the following error whilst compiling Cairo related libs:

```
  CC       images/bmp/libegt_la-cairo_bmp.lo
In file included from images/bmp/cairo_bmp.c:20:0:
/usr/arm-linux-gnueabi/include/stdlib.h:152:8: error: ‘_Float128’ is not supported on this target
 extern _Float128 strtof128 (const char *__restrict __nptr,
        ^~~~~~~~~
/usr/arm-linux-gnueabi/include/stdlib.h:164:8: error: ‘_Float64x’ is not supported on this target
 extern _Float64x strtof64x (const char *__restrict __nptr,
```

there are any workarounds?

joshua-henderson, 2020-06-02:
@amontefusco What's being compiled is a simple EGT source file that depends on cairo.  Not any cairo libs.

Your environment is a bit confusing and unknown.  Are you using a ubuntu armel multiarch setup for cross compiling (guessing based on your compiler path) or some other cross compiler?  What is your target processor?  Can you provide the summary at the end of the ./configure output?

At first glance, the issue is you are somehow mixing in host and target toolchains/headers/libs.  Because the error is in a toolchain header (stdlib.h), it looks like this has something to do with your environment.  This can happen, for instance, if you are using pkg-config that provides a host cairo package, but you are compiling for ARM.

You are compiling on ubuntu, for a debian target which is curious.  You need to make sure EGT is compiled against and links to the same shared libraries on your debian target (i.e. cairo, libjpeg, libdrm, libpng, libjpeg).  Are these ARM libraries installed/available in your cross compilation environment?  Usually they would be in /usr/lib/arm-linux-gnueabi/ based on where your stdlib.h is.

This is not a line for line tutorial, but something like this would be needed to get the dependent libs available and *then* compile and link on debian, for debian.  Note that not all EGT dependencies have debian packages (i.e. libplanes), so you would have to manually compile and make it available.

```
sudo apt install crossbuild-essential-armel
sudo dpkg --add-architecture armel
sudo apt update
sudo apt install libdrm-dev:armel libcairo2-dev:armel libjpeg-dev:armel libmagic-dev:armel libsndfile1-dev:armel libasound2-dev:armel libcurl4-openssl-dev:armel librsvg2-dev:armel
```

amontefusco, 2020-06-03:
Hi Joshua,
many thanks for taking the time to look at my question.

1. my target system is an AcmeSystems  Road Runner, a SOM based on SAMA5D27 Cortex-A5@500 MHz MPU .

Such system routinely runs Debian for ARM (after one builds a specific kernel as per AcmeSystems instructions) or Buildroot (there are specific patches to do that).  

2. I am attempting to build, on my laptop running U18.04, the EGT because compiling directly on the target system is painfully slow

3. I understand now it is a quite more complex process that I thought, even unfeasible on Ubuntu.

closed 2020-06-03 by amontefusco