linux4sam/egt #5: Local install of EGT not found by eclipse

michael-langford-carrier, 2020-06-04:
After following Eclipse IDE setup instructions (for local install of EGT, not cross compile), the link fails after each other step completed succesfully. It appears that even though libegt is installed correctly, and is found by the linker, the link fails, claiming there are undefined references. This makes no sense unless the libraries are completely corrupted.

`2:11:14 **** Incremental Build of configuration Build (GNU) for project egt-local ****
make all 
 cd . && /bin/bash /home/lab-pc-linux/eclipse-workspace/egt-local/missing automake-1.16 --foreign Makefile
 cd . && /bin/bash ./config.status Makefile depfiles
config.status: creating Makefile
config.status: executing depfiles commands
g++ -DPACKAGE_NAME=\"basic\" -DPACKAGE_TARNAME=\"basic\" -DPACKAGE_VERSION=\"1.0\" -DPACKAGE_STRING=\"basic\ 1.0\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -DPACKAGE=\"basic\" -DVERSION=\"1.0\" -I.  -pthread -I/usr/local/include -I/usr/include/gstreamer-1.0 -I/usr/include/orc-0.4 -I/usr/include/gstreamer-1.0 -I/usr/include/lua5.3 -I/usr/include/plplot -I/usr/include/x86_64-linux-gnu -I/usr/include/librsvg-2.0 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/cairo -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/pixman-1 -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/libpng16 -pthread -legt   -g -O2 -MT basic-basic.o -MD -MP -MF .deps/basic-basic.Tpo -c -o basic-basic.o `test -f 'basic.cpp' || echo './'`basic.cpp
mv -f .deps/basic-basic.Tpo .deps/basic-basic.Po
g++  -g -O2 -L/usr/local/lib -legt -lcairo -lpthread -pthread -legt  -o basic basic-basic.o  
/usr/bin/ld: basic-basic.o: in function `egt::v1::Button& egt::v1::center<egt::v1::Button>(egt::v1::Button&)':
/usr/local/include/egt/widgetflags.h:85: undefined reference to `egt::v1::Widget::align(egt::v1::Flags<egt::v1::AlignFlag> const&)'
/usr/bin/ld: basic-basic.o: in function `main':
/home/lab-pc-linux/eclipse-workspace/egt-local//basic.cpp:6: undefined reference to `egt::v1::Application::Application(int, char**, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)'
/usr/bin/ld: basic-basic.o: in function `egt::v1::Window::Window(egt::v1::PixelFormat, egt::v1::WindowHint)':
/usr/local/include/egt/window.h:69: undefined reference to `egt::v1::Window::DEFAULT_FORMAT'
/usr/bin/ld: /usr/local/include/egt/window.h:69: undefined reference to `egt::v1::Window::Window(egt::v1::RectType<int, (egt::v1::detail::Compatible)0> const&, egt::v1::PixelFormat, egt::v1::WindowHint)'
/usr/bin/ld: basic-basic.o: in function `egt::v1::TopWindow::TopWindow()':
/usr/local/include/egt/window.h:243: undefined reference to `vtable for egt::v1::TopWindow'
/usr/bin/ld: basic-basic.o: in function `main':
/home/lab-pc-linux/eclipse-workspace/egt-local//basic.cpp:9: undefined reference to `egt::v1::Button::default_text_align()'
/usr/bin/ld: /home/lab-pc-linux/eclipse-workspace/egt-local//basic.cpp:9: undefined reference to `egt::v1::Button::Button(egt::v1::Frame&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, egt::v1::Flags<egt::v1::AlignFlag> const&)'
/usr/bin/ld: /home/lab-pc-linux/eclipse-workspace/egt-local//basic.cpp:11: undefined reference to `egt::v1::Window::show()'
/usr/bin/ld: /home/lab-pc-linux/eclipse-workspace/egt-local//basic.cpp:13: undefined reference to `egt::v1::Application::run()'
/usr/bin/ld: /home/lab-pc-linux/eclipse-workspace/egt-local//basic.cpp:9: undefined reference to `egt::v1::Button::~Button()'
/usr/bin/ld: /home/lab-pc-linux/eclipse-workspace/egt-local//basic.cpp:8: undefined reference to `egt::v1::TopWindow::~TopWindow()'
/usr/bin/ld: /home/lab-pc-linux/eclipse-workspace/egt-local//basic.cpp:6: undefined reference to `egt::v1::Application::~Application()'
/usr/bin/ld: /home/lab-pc-linux/eclipse-workspace/egt-local//basic.cpp:9: undefined reference to `egt::v1::Button::~Button()'
/usr/bin/ld: /home/lab-pc-linux/eclipse-workspace/egt-local//basic.cpp:8: undefined reference to `egt::v1::TopWindow::~TopWindow()'
/usr/bin/ld: /home/lab-pc-linux/eclipse-workspace/egt-local//basic.cpp:6: undefined reference to `egt::v1::Application::~Application()'
collect2: error: ld returned 1 exit status
make: *** [Makefile:364: basic] Error 1
"make all" terminated with exit code 2. Build might be incomplete.

12:11:20 Build Failed. 21 errors, 0 warnings. (took 6s.472ms)
`

joshua-henderson, 2020-06-04:
@michael-langford-carrier It looks like you're not following the eclipse tutorial exactly, but what you're trying to do is perfectly fine.

The tutorial does indeed give you incorrect information.  If you look at your linker line, the -l flags are all *before* the input basic-basic.o.  That's not going to work in most cases - linker order does matter and the libs basic-basic.o depends on need to follow.  The problem starts with the Makefile.am.  Using basic_LDFLAGS is not the right variable to add the $(EGT_LIBS) to, and instead basic_LDADD should be used.  Change the line to:

`basic_LDADD = $(EGT_LIBS) -lpthread`

Make that change and hit build and you should be good to go.  I'll update this in the documentation.

michael-langford-carrier, 2020-06-04:
Fantastic - thank you, that worked perfectly! Never would have caught that.