linux4sam/egt #6: Cross Compiling

arahowerya, 2020-06-15:
Hi Guys,
The project looks great, and seems very impressive, have been looking at it for a few hours.

Something's a little unclear, I have a ATSAMA5D27-SOM1-EK board with a TM43xx LCD attached, and I can get the examples from the [linux4sam prebuilt yocto image demo archive](https://www.linux4sam.org/bin/view/Linux4SAM/Sama5d27Som1EKMainPage#Demo) to run on the screen - all looks great.

I can also make compile and run all the examples perfectly on a local ubuntu machine.

However, it's unclear how do I cross compile to the target? How do I end up taking for example the `basic.cpp` example and ending up with the `egt_basic` binary that's on the example yocto image?

Would appreciate any help
Kilian

noglitch, 2020-06-19:
There is a dedicated SDK on the page:
https://www.linux4sam.org/bin/view/Linux4SAM/SoftwareTools#Yocto_Project_SDK
For ATSAMA5D27-SOM1-EK, it's this one:
https://files.linux4sam.org/pub/demo/yocto-sdk/poky-atmel-glibc-x86_64-microchip-graphics-image-cortexa5t2hf-neon-vfpv4-sama5d2-xplained-sd-toolchain-3.1.sh
Install it, follow instruction on how to use it and EGT libraries must be present and ready to cross-compile.
Tell us if you find difficulties in the process.
Best regards,
  Nicolas

arahowerya, 2020-06-21:
Thanks for that, I've been using an old build of yocto for a while now, I'll try this out. Please consider pointing to that link in the README, I googled everything I could re: EGT and couldn't find that.
Thanks again for your help
Kilian

closed 2020-06-21 by arahowerya