linux4sam/egt #7: undefined reference to `egt::v1::Application::Application()

tcsabina, 2020-11-11:
Hi,

I am trying to setup my host system according to https://microchip.wikidot.com/32mpu:egt-eclipse.

egt is compiled and installed on the host (Linux Mint 19.2). All examples are working.

Sample project compilation in Eclipse stops at linking:
`00:49:48 **** Incremental Build of configuration Debug for project EgtProject ****
make all 
Building file: ../basic.cpp
Invoking: Cross G++ Compiler
arm-buildroot-linux-gnueabi-g++ -I/usr/local/include -I/usr/include/cairo -O0 -g3 -Wall -c -fmessage-length=0 `pkg-config libegt –cflags` -MMD -MP -MF"basic.d" -MT"basic.o" -o "basic.o" "../basic.cpp"
arm-buildroot-linux-gnueabi-g++: WARNING: unsafe header/library path used in cross-compilation: '-I/usr/local/include'
arm-buildroot-linux-gnueabi-g++: WARNING: unsafe header/library path used in cross-compilation: '-I/usr/include/cairo'
Finished building: ../basic.cpp
 
Building target: EgtProject
Invoking: Cross G++ Linker
arm-buildroot-linux-gnueabi-g++ -L/usr/local/lib `pkg-config libegt —libs` -o "EgtProject"  ./basic.o   -lpthread
arm-buildroot-linux-gnueabi-g++: WARNING: unsafe header/library path used in cross-compilation: '-L/usr/local/lib'
/home/toma/workspace/buildroot-at91/output/host/lib/gcc/arm-buildroot-linux-gnueabi/8.4.0/../../../../arm-buildroot-linux-gnueabi/bin/ld: warning: library search path "/usr/local/lib" is unsafe for cross-compilation
/home/toma/workspace/buildroot-at91/output/host/lib/gcc/arm-buildroot-linux-gnueabi/8.4.0/../../../../arm-buildroot-linux-gnueabi/bin/ld: ./basic.o: in function `main':
/home/toma/workspace/EgtProject/Debug/../basic.cpp:14: undefined reference to `egt::v1::Application::Application(int, char**, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)'
/home/toma/workspace/buildroot-at91/output/host/lib/gcc/arm-buildroot-linux-gnueabi/8.4.0/../../../../arm-buildroot-linux-gnueabi/bin/ld: /home/toma/workspace/EgtProject/Debug/../basic.cpp:14: undefined reference to `egt::v1::Application::~Application()'
makefile:44: recipe for target 'EgtProject' failed
collect2: error: ld returned 1 exit status
make: *** [EgtProject] Error 1
"make all" terminated with exit code 2. Build might be incomplete.

00:49:49 Build Failed. 5 errors, 1 warnings. (took 1s.716ms)

`

I have seen #5 , but I am not really able to follow the suggestion in there. It seems that there has been some changes in the tutorial, as we need to use pkg-config, (which is not visible in #5 ), and there is no Makefile.am anymore.

ldesroches, 2020-11-13:
Hi,

I am not used to eclipse but I see several suspicious things:
- it seems you are using the '—' character instead of '--' for --cflags and --libs
- you are giving some include paths related to your host machine, this is why you have some warnings
- contrary to the tutorial, I would put `pkg-config libegt --libs` in Cross G++ Linker -> Miscellaneous ->  Linker flags. I don't know how eclipse behaves but I fear having it in Libraries may prevent the expansion of the pkg command and may add an extra -l.

Hope it helps

Regards

tcsabina, 2020-11-16:
Hi,

Thanks so much for the reply.
I was able to solve this base on your instructions. The problem was indeed the '—' character instead of '--'.

Here again the code snippet from the tutorial again:
`pkg-config libegt —libs`

This is a pure copy+paste, so this is wrong on the tutorial page.
And someone like me, less familiar with the usage an command line parameters of pkg-config, can easily overlook this.

Is there a way to let the page be updated?

Regards,
Tamas

ldesroches, 2020-11-16:
Sure, I will report the issue.

closed 2020-11-17 by tcsabina