linux4sam/egt #8: undefined reference to `egt::v1::Widget::parent_in_layout()'

tcsabina, 2020-11-16:
Hi,

I am at Exercise #1 (https://microchip.wikidot.com/32mpu:egt-exercise-solution-1) of the EGT tutorial, but I have the following problem:

> 21:39:26 **** Incremental Build of configuration Debug for project EgtProject ****
make all 
Building file: ../basic.cpp
Invoking: Cross G++ Compiler
arm-buildroot-linux-gnueabi-g++ -I/usr/local/include -I/usr/include/cairo -O0 -g3 -Wall -c -fmessage-length=0 `pkg-config libegt --cflags` -MMD -MP -MF"basic.d" -MT"basic.o" -o "basic.o" "../basic.cpp"
arm-buildroot-linux-gnueabi-g++: WARNING: unsafe header/library path used in cross-compilation: '-I/usr/local/include'
arm-buildroot-linux-gnueabi-g++: WARNING: unsafe header/library path used in cross-compilation: '-I/usr/include/cairo'
Finished building: ../basic.cpp
 
Building target: EgtProject
Invoking: Cross G++ Linker
arm-buildroot-linux-gnueabi-g++ `pkg-config libegt --libs` -o "EgtProject"  ./basic.o   -lpthread
/home/toma/workspace/buildroot-at91/output/host/lib/gcc/arm-buildroot-linux-gnueabi/8.4.0/../../../../arm-buildroot-linux-gnueabi/bin/ld: ./basic.o: in function `egt::v1::Widget::width(int)':
/usr/local/include/egt/widget.h:293: undefined reference to `egt::v1::Widget::parent_in_layout()'
makefile:44: recipe for target 'EgtProject' failed
collect2: error: ld returned 1 exit status
make: *** [EgtProject] Error 1
"make all" terminated with exit code 2. Build might be incomplete.

21:39:28 Build Failed. 4 errors, 0 warnings. (took 1s.790ms)

The code is this:
`#include <egt/ui>
#include <iostream>

using namespace std;
using namespace egt;

int main(int argc, const char ** argv)
{
  Application app;
  TopWindow window;
  Button button(window, "Press Me");

//  center(button);
//  button.align(AlignFlag::center | AlignFlag::left);

  button.move(Point(300,300));
  cout << "X:" << button.x() << " Y:" << button.y() << endl;

  //change widget width and height
  button.width(button.width()*2);
//  button.height(70);
  cout << "width:" << button.width() << " height:" << button.height() << endl;

  window.show();

  return app.run();
}
`
The exercise suggesting that the 1st line should be this:
`#include <egt ui="">`
But I have never seen anything like that in a c(pp) source file. So I am using a 'normal' include:
`#include <egt/ui>
`

Why do I have that unresolved symbol?

ldesroches, 2020-11-16:
Hi,
Right `#include <egt ui="">` is wrong. I will report this error.

Concerning your linking problem, I don't reproduce it. Do you have also EGT installed on your PC? I tend to think you have a recent version installed on your PC and an older one within Buildroot.
If it's the case, there is an issue with your Eclipse setup. It takes the EGT library from your PC instead of the one from Buildroot.

Regards

tcsabina, 2020-11-17:
Hi,

There is indeed an EGT install on my PC. I followed the guide: https://microchip.wikidot.com/32mpu:egt-prepare#prepHost.
Step 3 says
configure
make
make install

So the EGT version from github is 'installed' on the PC (make install copies binaries and libraries to /usr/local/, so it is indeed visible).

There is probably one in buildroot as well. In the buildroot-external, if I am not mistaken.
How shall I configure my PC to 'use' the EGT library from buildroot? Or at least tell the Eclipse project to use it.

Tamas

tcsabina, 2020-11-17:
oh wait,
I've just realized that I am using buildroot from linux4sam: https://github.com/linux4sam/buildroot-at91.git
instead of the one mentioned in the tutorial of EGT: https://git.buildroot.net/buildroot

let me do some testing with the latter...

ldesroches, 2020-11-18:
Hi,
It won't solve your issue. The problem is with your eclipse configuration.
The root cause of your problem is probably here:
```
arm-buildroot-linux-gnueabi-g++ -I/usr/local/include -I/usr/include/cairo -O0 -g3 -Wall -c -fmessage-length=0 pkg-config libegt --cflags -MMD -MP -MF"basic.d" -MT"basic.o" -o "basic.o" "../basic.cpp"
arm-buildroot-linux-gnueabi-g++: WARNING: unsafe header/library path used in cross-compilation: '-I/usr/local/include'
arm-buildroot-linux-gnueabi-g++: WARNING: unsafe header/library path used in cross-compilation: '-I/usr/include/cairo'
```
You should not have '-I/usr/local/include -I/usr/include/cairo'. Check the properties of your project. I assume there is some wrong in the Cross G++ Compiler configuration. Maybe in the Includes or Miscellaneous section.

Here is the compilation logs I have:
```
Building file: ../basic.cpp
Invoking: Cross G++ Compiler
arm-buildroot-linux-gnueabihf-g++ -O0 -g3 -Wall -c -fmessage-length=0 `pkg-config libegt --cflags` -MMD -MP -MF"basic.d" -MT"basic.o" -o "basic.o" "../basic.cpp"
Finished building: ../basic.cpp
```

Regards

tcsabina, 2020-12-14:
Hi ldesroches,

Sorry for the late reply. I had no chance to play around with this...

You solved the problem! Indeed those path entries were manually added by me to the Eclipse project. For some reason the indexer was not able to find the correct definitions for the variable types, so I've added these path entries to the project.

After I removed them, I was able to compile and link!
And on top of this, Eclipse's indexer is also happy now.

Thanks and regards,
Tamas

closed 2020-12-14 by tcsabina

ldesroches, 2020-12-15:
Hi Tamas,

Thanks for the feedback.

Regards

Ludovic