linux4sam/egt #10: Build EGT error

NelsonKon, 2021-02-05:
Hi,
I follow these steps, 
git clone --recursive https://github.com/linux4sam/egt.git
cd egt
./autogen.sh
./configure
When I run make, I got this error,
buildroot-at91/output/host/arm-buildroot-linux-gnueabihf/sysroot/usr/include/gnu/stubs.h:7:11: fatal error: gnu/stubs-soft.h: No such file or directory
    7 | # include <gnu/stubs-soft.h>
      |           ^~~~~~~~~~~~~~~~~~

I don't know it solves the problem, but I just created an empty file called stubs-soft.h in the gnu directory.  When I run make, the problem went away.  Another error came in, 

/usr/include/c++/9/memory: In function ‘void* std::align(std::size_t, std::size_t, void*&, std::size_t&)’:
/usr/include/c++/9/memory:121:58: error: cast from ‘void*’ to ‘uintptr_t’ {aka ‘unsigned int’} loses precision [-fpermissive]
  121 |   const auto __intptr = reinterpret_cast<uintptr_t>(__ptr);
      |                                                                                          ^

Do you know what went wrong?

Hope to hear from you soon.

Regards,
Nelson

ldesroches, 2021-02-05:
Hi Nelson,

What are you trying to do? Cross-compile EGT manually? If yes, there are missing options when you call configure. Why don't you let Buildroot compiling EGT? The package is in buildroot-external-microchip.

Regards

Ludovic

NelsonKon, 2021-02-05:
HI Ludovic,
Thanks for coming back.

I did that because I can't load my application program using eclipse.  I got the error of unable to create primary plane.  I am thinking if I can build the egt, build the application program and copy over to the SD card.  Can I do it that way?

Hope to hear from you soon.

Regards,
Nelson

NelsonKon, 2021-02-05:
Hi Ludovic,
What configure option you meant?

Regards,
Nelson

closed 2021-02-08 by ldesroches

SandeepSM, 2021-02-16:
Hi Nelson, 
      For cross compilation of egt. please refer the steps mentioned in egt/docs/src/started.md file.

Regards,
Sandeep Sheriker M

NelsonKon, 2021-02-17:
Hi Sandeep,
Thanks for coming back.
I followed the steps in the egt/docs/src/started.md.  

But I am still getting this error,
~/linux4sam/egt$ make
make  all-recursive
make[1]: Entering directory '/home/nelsonkon/linux4sam/egt'
Making all in external
make[2]: Entering directory '/home/nelsonkon/linux4sam/egt/external'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/nelsonkon/linux4sam/egt/external'
Making all in src
make[2]: Entering directory '/home/nelsonkon/linux4sam/egt/src'
make  all-am
make[3]: Entering directory '/home/nelsonkon/linux4sam/egt/src'
  CXX      libegt_la-animation.lo
In file included from ../include/egt/detail/meta.h:18,
                 from ../include/egt/animation.h:15,
                 from animation.cpp:7:
/usr/include/c++/9/memory: In function ‘void* std::align(std::size_t, std::size_t, void*&, std::size_t&)’:
/usr/include/c++/9/memory:121:58: error: cast from ‘void*’ to ‘uintptr_t’ {aka ‘unsigned int’} loses precision [-fpermissive]
  121 |   const auto __intptr = reinterpret_cast<uintptr_t>(__ptr);
         |                                                                                        ^
In file included from /home/nelsonkon/linux4sam/buildroot-at91/output/host/arm-buildroot-linux-gnueabihf/sysroot/usr/include/bits/fcntl.h:54,
                 from /home/nelsonkon/linux4sam/buildroot-at91/output/host/arm-buildroot-linux-gnueabihf/sysroot/usr/include/fcntl.h:35,
                 from ../include/egt/asio/detail/socket_types.hpp:70,
                 from ../include/egt/asio/impl/error_code.ipp:29,
                 from ../include/egt/asio/error_code.hpp:201,
                 from ../include/egt/asio/detail/scheduler_operation.hpp:18,
                 from ../include/egt/asio/detail/executor_op.hpp:22,
                 from ../include/egt/asio/impl/system_executor.hpp:18,
                 from ../include/egt/asio/system_executor.hpp:129,
                 from ../include/egt/asio/associated_executor.hpp:21,
                 from ../include/egt/asio.hpp:21,
                 from ../include/egt/timer.h:15,
                 from ../include/egt/animation.h:18,
                 from animation.cpp:7:
/home/nelsonkon/linux4sam/buildroot-at91/output/host/arm-buildroot-linux-gnueabihf/sysroot/usr/include/bits/fcntl-linux.h: At global scope:
/home/nelsonkon/linux4sam/buildroot-at91/output/host/arm-buildroot-linux-gnueabihf/sysroot/usr/include/bits/fcntl-linux.h:357:26: warning: ISO C++ forbids zero-size array ‘f_handle’ [-Wpedantic]
  357 |   unsigned char f_handle[0];
         |                                         ^
make[3]: *** [Makefile:2456: libegt_la-animation.lo] Error 1
make[3]: Leaving directory '/home/nelsonkon/linux4sam/egt/src'
make[2]: *** [Makefile:1978: all] Error 2
make[2]: Leaving directory '/home/nelsonkon/linux4sam/egt/src'
make[1]: *** [Makefile:1449: all-recursive] Error 1
make[1]: Leaving directory '/home/nelsonkon/linux4sam/egt'
make: *** [Makefile:1350: all] Error 2

What is the error, actually?

SandeepSM, 2021-02-17:
Hi Nelson,
       Looks like cross compiler path is not set correctly. can you follow below step and check ?

export PATH=<buildroot-at91>/output/host/bin:$PATH
./autogen.sh  
./configure --prefix=/tmp/ --host=arm-buildroot-linux-gnueabihf
make 
make install 

Regards,
Sandeep Sheriker M

NelsonKon, 2021-02-18:
Hi Sandeep,
Thanks a lot for your help.  It works.

But at the configure option, I don't use the --prefix.  I wonder what it is.

Anyway, I manage to run make till the end without any error.

Thanks so much.

Regards,
Nelson