linux4sam/egt #9: EGT primary plane

NelsonKon, 2021-02-05:
Hi, 
I am using ATSAMA5D27-SOM1-EK1 board and build using buildroot with sama5d27_som1_ek_graphics_defconfig.  After booting up, I use eclipse to create the basic program and tried to run on the board.  But I got this error message,

terminate called after throwing an instance of 'std::runtime_error'
  what():  unable to create primary plane
Aborted

Any idea how to solve it?

What is actually the primary plane?  How to create it?

ldesroches, 2021-02-05:
Hi,

Usually this error happens when you already have an instance of EGT running. The primary plane is the base layer of the LCD controller and it's probably already used by another EGT application.

Regards

Ludovic

NelsonKon, 2021-02-05:
Hi Ludovic,
Thanks for coming back. 
So, I am not supposed to load the board to the LCD while I load my application program using Eclipse to the board?

How can I do that?

Regards,
Nelson

ldesroches, 2021-02-05:
Before loading the application, the demo must be stopped '/etc/init.d/S99demo stop', double check with 'ps' there is any EGT application running, otherwise kill it.

Regards

Ludovic

NelsonKon, 2021-02-08:
Hi Ludovic,
After I stopped the S99demo, I am able to load my program created using eclipse.  But after that when I start the S99demo again, my program actually blocks it.  I can't close my program also.  
Is there a way to do button action?  I am thinking of after I press the button, I can so some other thing, like closing the window, etc.
Regards,
Nelson

ldesroches, 2021-02-08:
Hi Nelson,
Kill the application if you can't manually close it. I don't use Eclipse, but I assumed there is a way to stop the application it runs. If you want to stop your application when you press a button, you have to handle it in your app and call App::quit(). The 'keys' example show you how to deal with input events.

As I see nothing wrong with EGT, I am closing your issues. You can ask for help on the EGT forum: https://www.microchip.com/forums/f551.aspx

Regards,
Ludovic

closed 2021-02-08 by ldesroches