linux4sam/egt #13: Segmentation Fault

stevenls25, 2021-03-05:
When running your dialogs program, It compiles ok and starts ok, I see all the buttons,
but the File Open and Message Dialog buttons give me a Segmentation fault on my board 
(with SAMA5D27 SOM1).  The other three buttons seem to work properly.

[image]

SandeepSM, 2021-03-09:
Hi,
   I am not able to reproduce segmentation fault you observed from out egt_dialog application. can you be more specific or the steps to reproduce this issue. if possible can you share your code snippet of dialog_box_demo ?

Regards,
Sandeep Sheriker M

stevenls25, 2021-03-09:
Was hardware issue not software.
Thank You

closed 2021-03-09 by stevenls25