linux4sam/egt #16: ScrolledView Issue

hoseck, 2021-04-15:
If I change the ScrollWheel on the ScrolledView, the contents look overlapping as shown in the figure.
[image]

ldesroches, 2021-04-16:
Hi,

Thanks for the report. We'll have a look.

Regards,
Ludovic

ldesroches, 2021-04-23:
Hi @hoseck 

Fix pushed. Let me know if's ok and if it doesn't have nasty side effects for your app.

Regards,
Ludovic

hoseck, 2021-04-30:
This bug still hasn't been fixed.
ScrollWheel Widget works well on the Notebook Tab. As you can see on the screen, there is a problem that text overlaps in ScrollView.
Please check again.

ldesroches, 2021-05-04:
Hi @hoseck 

I get a notification where you said it was not solved but there is no comment here. Did you delete it? Is it solved?

If it's not solved with the master branch, is there a way you share your code. I was able to reproduce the issue and it's fixed on my side but maybe some cases are not handled correctly.

Regards,
Ludovic

closed 2021-05-12 by ldesroches