linux4sam/egt #18: How to prevent libinput from capturing input device?

OOHehir, 2022-02-10:
**Summary**: libinput is the default method for EGT of capturing inputs from devices but how does one prevent it capturing inputs from say /dev/event0 ? 

Hello,
I've [recently worked](https://www.microchip.com/forums/FindPost/1198777) to add input from a resistive touchscreen input to a SAMA5D4 Xplained board via the on chip ADC. I calibrated the screen using the tslib tool ts_calibrate & ran the ts_uinput script which creates another event device for a GUI framework to capture. I found when I used this that the touchscreen performance was unsatisfactory. It appears that EGT was still capturing input from the raw inputs of the touchscreen as well as the calibrated input from  ts_uinput. 

I temporarily solved the issue by removing the raw input (i.e. rm /dev/event0) which resulted in satisfactory touchscreen input. 

Is there a environmental option to prevent libinput capturing an event device? I've taken a look through the documents but can't see any.

Regards,

Owen

ldesroches, 2022-02-18:
Hi,
No there is no way to disable libinput from environment variable. This is the default input for EGT, so if an input device is handled by the libinput, it will be used.
If you want to avoid this situation, you can build EGT without the libinput support or you have to tune Application::setup_inputs() in src/app.cpp. Move m_inputs.push_back(std::make_unique<detail::InputLibInput>(*this)); somewhere else depending on the behavior you expect.

Regards,
Ludovic

OOHehir, 2022-02-22:
OK,

I think I'll come up with a startup script to solve the issue.

closed 2022-02-22 by OOHehir