linux4sam/egt #20: Calc Example Issue

swsuh66, 2022-03-28:
We have used egt 1.1 for our project. Now we migrated to egt 1.3 version.
It seems everything is OK except Calc example. Our project use calc example fot various input situations 

In EGT 1.3, digit input has some issue. The input shows only for last two digits. But the result is OK.

Thanks
Regards
Brian

ldesroches, 2022-05-02:
Hello Brian,

It will be fixed in 1.3.1 and 1.4. You can checkout the 1.3 branch to get the fix right now.
The cause of this issue is a performance optimization with textbox which didn't handle non left alignment.

Regards,
Ludovic

swsuh66, 2022-05-03:
Dear Ludovic,

Numeric input seems to work well.
But "C"lear key still have problem. Nothing cleared.

Thanks & Regards,
Brian

ldesroches, 2022-05-03:
Hello Brian,

Thanks for the feedback, we didn't notice the issue with the clear method ot the textbox. It's fixed, 1.3.1 is released.

Regards,
Ludovic

swsuh66, 2022-05-04:
Dear Ludovic,

Thanks, It works well

Regards
Brian

closed 2022-05-04 by ldesroches