linux4sam/egt #21: ScrolledView issue

OhYoungJin1, 2022-05-06:
Scroller background not transparent in ScrolledView

I found 

[view.cpp 102]
To prevent this situation, we force the background drawing of the ScrolledView even if fill flags are set to none.

please...

ldesroches, 2022-05-10:
Hi,

Sorry I don't get the issue. You have an issue with the horizontal and vertical scrollers?
[image]
Is something wrong with this?

Can you share a picture of what is not behaving as you expect?

Regards,
Ludovic

OhYoungJin1, 2022-05-16:
[image]

I want the gray area to be transparent

ex)
auto list = make_shared<egt::ScrolledView>(ScrolledView::Policy::as_needed, ScrolledView::Policy::as_needed);
list->fill_flags(Theme::FillFlag::blend);
list->color(Palette::ColorId::bg, Palette::transparent);
-----> 
[image]

auto list = make_shared<egt::ScrolledView>(ScrolledView::Policy::as_needed, ScrolledView::Policy::as_needed);
list->fill_flags(Theme::FillFlag::blend);
list->color(Palette::ColorId::bg, egt::Color(229, 229, 229));
-----> 
[image]

list->color(Palette::ColorId::bg, Palette::transparent);  --> background color is black!
list->color(Palette::ColorId::bg, egt::Color(229, 229, 229));  --> background color is gray!

ldesroches, 2022-05-31:
The issue is probably related to the widgets inside your ScrolledView

egt::TopWindow window;
window.color(egt::Palette::ColorId::bg, egt::Palette::blue);
egt::ScrolledView view(window, egt::Rect(0, 0, 200, 200));
egt::Frame frame(egt::Rect(egt::Point(), egt::Size(400, 400)));
frame.fill_flags(egt::Theme::FillFlag::blend);
frame.color(egt::Palette::ColorId::bg, egt::Palette::transparent);
view.add(frame);
window.show();

[image]

OhYoungJin1, 2022-06-24:
The issue is probably related to the widgets inside your ScrolledView
------> Everything was resolved well !!

Thank you!!!

closed 2022-06-24 by OhYoungJin1