linux4sam/egt #30: Does EGT support the SAMA7D65 CPU2D

kay54068, 2025-01-11:
Hello, may I ask if EGT supports the CPU2D of the SAMA7D65?
I am not referring to the GFX2D of the SAM9X60 series.

ldesroches, 2025-01-13:
Hello,

No, at the moment, none of the 2D GPUs are supported. The GFX2D support should be added this year. Concerning the GPU of the SAMA7D65, we are working on it, but we have no schedule to communicate yet.

Regards,
Ludovic

kay54068, 2025-01-13:
@ldesroches 

That's truly unfortunate. Currently, our company has started using the SAMA7D65 and requires a GPU to assist in accelerating video playback to reduce CPU usage. 

Are there any alternative solutions, such as libm2d?
https://github.com/linux4sam/libm2d/tree/master

Or Is there a plan to integrate the SAMA7D65 CPU2D into the Linux kernel to support OpenGL ES?

CyrillePitchen, 2025-01-13:
> ADRXVZF2VCBY5GRE7ERR64T2KPD5RAVCNFSM6AAAAABVACPQKWVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDKOBXGIZTQOJSGI>.

Hi,

> EXTERNAL EMAIL: Do not click links or open attachments unless you know 
> the content is safe
> 
> @ldesroches <https://github.com/ldesroches>
> 
> That's truly unfortunate. Currently, our company has started using the 
> SAMA7D65 and requires a GPU to assist in accelerating video playback to 
> reduce CPU usage.

The Vivante GC520UL 2D GPU embedded within the Microchip SAMA7D65 SoC is 
only capable of operating on rasters like drawing rectangles, pictures, 
blending them but AFAIK, it has no video decoder capabilities. The "UL" 
stands for "Ultra Light", therefore even lighter than the GC520L 2D GPU 
that you could find on other SoCs.

> Are there any alternative solutions, such as libm2d?
> https://github.com/linux4sam/libm2d/tree/master <https://github.com/ 
> linux4sam/libm2d/tree/master>

libm2d is being reworked to abstract hardware GPUs (currently the GFX2D, 
later the GC520UL) and to provide user applications (like those built 
upon libegt) a common API to access the GPU but only to perform 
operations on rasters, such as drawing rectangles with or without a 
single texture/image (otherwise a flat color is used) and with or 
without blending (alpha blending, or other types of blending).
Hence, this new API is inspired from OpenGL to setup blending parameters 
but limited to very few operations.

However, there won't be any video decoder feature since the targeted 
pieces of hardware don't support it.

> Or Is there a plan to integrate the SAMA7D65 CPU2D into the Linux kernel 
> to support OpenGL ES?

OpenGL ES is a subset or simplified version of OpenGL, which is a 3D 
library: it works on vector images.

On the other hand the Vivante GC520UL is a 2D GPU so it can work only 
bitmap (raster). So at best it could implement a very small subset of 
OpenGL ES, and this is what we do in the WIP libm2d.

BR,

Cyrille

closed 2025-03-19 by ldesroches