linux4sam/egt #33: Memory leak with EGT demo?

gbmbg, 2025-04-16:
Good Morning,
We encountered  memory leak issues while implementing code using the EGT libraries. To ensure that the problem was not specific to our code, we tested the same scenario with the demo provided by the EGT libraries. Unfortunately, the memory leak persisted, indicating that the issue may lie within the libraries themselves.
In fact, flashing the NAND with the demo on SAM9x60-curiosity board, after 25/26 hour we encountered the same problem pointed out in  #31, the EGT demo process was killed by the OOM-killer:

[image]

[image]

[image]

Do you have any suggestion regarding this? We are using 2024.10 buildroot based kernel
Thanks a lot and work well

ldesroches, 2025-04-17:
Hello,

This sounds quite different from the other issue that was about caching more and more images. Right now, we didn't observe it, but we didn't test it on such duration.

We'll have a look as soon as possible.

Thanks for reporting this issue.

Regards,
Ludovic

gbmbg, 2025-04-17:
Ok thanks a lot, from our experience and our code issues, it seems connected to animations... 
Hope this helps.

ldesroches, 2025-04-22:
Hello,

We have not started the debugging yet. Thanks for your inputs. For sure, it will be useful.

Regards,
Ludovic

gbmbg, 2025-04-30:
Hello do you have any news?

ldesroches, 2025-05-02:
Hello,

Sorry, no, not yet. We are in the release process of 1.11, we'll have a look once the release is done. As it is sync with linux4microchip we can't postpone it. Depending of the fix, we'll release a 1.11.x or it will be in the 1.12.

Regards,
Ludovic

gbmbg, 2025-05-02:
Ok, so we will have to wait about 3/4 month at minimum, right?
Please let me know
Work well
Giovanni

ldesroches, 2025-05-02:
I can't tell until we find the root cause. It can be due to EGT or other components used by EGT and even the LCD kernel driver.

If it's related to EGT and if the fix is trivial, i.e. we don't expect side effects, we can release a 1.11.x so it could be done as soon as we find the fix. If it's more complex, it will be put on the master branch so you could cherry-pick it. It just means that the patch hasn’t yet passed all the tests we run before a release.

We still have issues to fix for 1.11, I don't think we'll be able to really dig into this issue before June.

gbmbg, 2025-05-02:
Ok thaks a lot.

ldesroches, 2025-05-06:
Hello,

We investigated more on the issue. We have not found any memory leak in process of the EGT application.
There is a memory leak with the slab allocator that probably comes from the LCD driver. I know there are plans to refresh this driver and so it can be part of this rework. It means that you'll probably have to wait for the next linux4microchip release.
I'll keep you informed when patches will be published, either for just fixing this bug, or when the driver refresh is done.

Regards,
Ludovic

gbmbg, 2025-05-06:
Ok thanks a lot, you mean at least 2025.10 right?

ldesroches, 2025-05-06:
yes

gbmbg, 2025-05-22:
Is EGT version 1.11 still based on cairo? If so, on which version?
Cairo v 1.17.4 is the one on which EGT v 1.10 is based, correct?

gbmbg, 2025-05-22:
Thanks a lot.
Now we are still using EGT 1.10 embedded in linux4microchip-2024.10 and it is based on 1.17.4!
looking to:
https://www.cairographics.org/news/[cairo-1.17.8](https://www.cairographics.org/news/cairo-1.17.8/)/
and
https://www.cairographics.org/news/[cairo-1.18.0](https://www.cairographics.org/news/cairo-1.18.0/)/
they fixed a lot of memory leaks!
Could be our issue related to theirs?
Thaks again and work well!

ldesroches, 2025-07-18:
Hello @bblessing1180 

These two patches have been tested on top on the latest linux4microchip release, i.e. 2025.04. They would be included in 2025.10.

Regards,
Ludovic

bblessing1180, 2025-07-18:
Hello @ldesroches 

Thanks! I updated the patch to work with 2025.04, and it has solved my problem as well. I truly appreciate this!

`diff --git a/drivers/gpu/drm/atmel-hlcdc/atmel_hlcdc_plane.c b/drivers/gpu/drm/atmel-hlcdc/atmel_hlcdc_plane.c
index 22f8199babba..3b997eeb57b3 100644
--- a/drivers/gpu/drm/atmel-hlcdc/atmel_hlcdc_plane.c
+++ b/drivers/gpu/drm/atmel-hlcdc/atmel_hlcdc_plane.c
@@ -1201,8 +1201,7 @@ atmel_hlcdc_plane_atomic_duplicate_state(struct drm_plane *p)
                return NULL;
        }

-       if (copy->base.fb)
-               drm_framebuffer_get(copy->base.fb);
+       __drm_atomic_helper_plane_duplicate_state(p, &copy->base);

        return &copy->base;
 }
@@ -1220,8 +1219,7 @@ static void atmel_hlcdc_plane_atomic_destroy_state(struct drm_plane *p,
                              state->dscrs[i]->self);
        }

-       if (s->fb)
-               drm_framebuffer_put(s->fb);
+       __drm_atomic_helper_plane_destroy_state(s);

        kfree(state);
 }
`