linux4sam/egt #34: Image Cache Optimization for Limited RAM Devices

mkumar7404508129, 2025-05-14:
Background
While using branch 1.10, I encountered a memory issue when dealing with large image sets (e.g., >500 images). The current implementation of image loading in egt/src/detail/imagecache.cpp uses a cache mechanism that does not enforce any upper limit on the number of images stored. This can result in a hard fault on devices with low RAM, as memory usage grows unchecked.

Issue
Unbounded Image Caching: The cache does not limit how many images can be stored, which causes memory exhaustion.

No Cache Eviction Policy: Once images are added, there is no mechanism to discard unused or least recently used images from the cache.

Solution
I made the following improvements in egt/src/detail/imagecache.cpp:

Added a configurable maximum cache size (e.g., via a constant or runtime config).

Implemented a simple Least Recently Used (LRU) eviction policy to discard old images when the cache limit is reached.

Provided a function to explicitly discard images from the cache, to allow manual cache control when needed.

Key Code Changes
Added a constexpr size_t kMaxCacheSize = 100; limit (can be made configurable).

Modified the image insertion logic to check the cache size and evict the least recently used image.

Introduced discard(const std::string& key) to remove specific images from the cache.

Impact
These changes:

Prevent hard faults on low-RAM systems.

Improve cache reliability and performance.

Provide developers with manual control over cache memory usage.

Recommendation
It is highly recommended to integrate this change in the mainline to avoid similar issues for other developers and devices with limited memory.

Please check  for the reference
[imagecache.txt](https://github.com/user-attachments/files/20199849/imagecache.txt)

gbmbg, 2025-05-22:
Good morning 
Did you face something like us (https://github.com/linux4sam/egt/issues/33) before using your improvement?
And if yes, did it solve your problem? Is your RAM usage stable now? 
Best Regards

mkumar7404508129, 2025-05-23:
Previously, my app was crashing due to memory issues—specifically from excessive caching. The crash used to happen within 2 minutes of continuous use and consistently triggered the same error you referenced (#33).

After optimising the image buffer to hold only 5 images at a time, the issue has been resolved. I tested the app continuously for 5 hours after this change, and it has been running smoothly without any crashes.